TitaniumTigers4829/mako-robot-code-2025 #6: Write swerve x stance code

Ishan1522, 2025-01-07:
When the robot has stopped for a couple seconds the modules should rotate to form an X. This makes it harder to push the robot around.

JacksonElia, 2025-01-07:
I don't think this should be when you stop for a few seconds, it should be while the driver is holding a button. That way they can easily & consciously activate/deactive it.

Ishan1522, 2025-01-07:
that is completely up to the driver. 
 
however imo we should be making it as easy to drive as possible. does giving them another button to worry about rlly help them?

Ishan1522, 2025-01-07:
Also @JacksonElia this is @Abigail-27 's project so i dont rlly want to be making any hard decisions here, since ultimately she is leading (and learning how to write) this.

JacksonElia, 2025-01-07:
Fair enough, this is a good project to get comfy with swerve for sure 🫡

closed 2025-01-16 by Ishan1522