TitaniumTigers4829/mako-robot-code-2025 #13: Configure Limelights

JacksonElia, 2025-01-08:
Pretty soon new limelight drivers will release, and we need to update the limelight when that happens. https://docs.limelightvision.io/docs/resources/downloads

When we do this, we will need to practice configuring settings and tuning the limelight. **Also we have to make sure we configure the acceptable april tag IDs (1-22) as we don't want the limelight detecting a non existent one (E.g. april tag 587).**

While tuning, as per the docs "Use the lowest exposure possible, and increase black level offset until field lights and LED reflections are removed from the image." We (in the slack) have a profile already tuned, but we should still practice anyways.

Ishan1522, 2025-01-08:
Very important

Ishan1522, 2025-01-08:
This is also important to understand bc we will moderately retune during comp

JacksonElia, 2025-01-09:
He posted that 2025.0 will be released on the 15th, so we can do this then

closed 2025-02-01 by Ishan1522