TitaniumTigers4829/mako-robot-code-2025 #19: Test if the filtering works

Ishan1522, 2025-01-09:
This is kinda just a note for the future, but I want to test (and tune) if the filtering I wrote with the discrepancy and distance stuff actually works. 
[image]
We know everything other than the discrepancy and distance checks work, so we just need to see if these are actually doing their job (I'm kinda sure that they do, but I didn't test this as much, so I think we should revisit this while tuning.

JacksonElia, 2025-01-09:
Yeah maybe on saturday if we have time we can try to really iron out a lot of these values. Though maybe the 2025.0 will like improve stuff and we should wait until then.

Ishan1522, 2025-01-10:
No saturday meeting 😭 I don't think anything will be set up (or enough of things will be set up) by wednesday for us to test this.

closed 2025-02-01 by Ishan1522