TitaniumTigers4829/mako-robot-code-2025 #36: Make canSeeAprilTags work from just Network Tables

JacksonElia, 2025-01-18:
If we choose to use `canSeeAprilTags` within PhysicalVision, we have to make it so it is based just on the network table entries rather than using limelight helpers/parsing the json.

Even if we stop using it for the pose updates, we still use it for this method.
```java
  @Override
  public double getLimelightAprilTagDistance(Limelight limelight) {
    if (canSeeAprilTags(limelight)) {
      return limelightEstimates[limelight.getId()].avgTagDist;
    }
    // To be safe returns a big distance from the april tags if it can't see any
    return Double.MAX_VALUE;
  }
```

The alternative to this would be figuring out what .avgTagDist returns when it doesn't see any april tags and make it return `Double.MAX_VALUE` instead. I would rather make it based on network table entries so we don't run into this problem again. I think you could do this by finding what values are returned by tx and ty if it doesn't see any april tags (maybe its like 0 or null or smth).

Also let's at warning comments (or maybe just one near the top of the file) warning about not creating circular logic.

Ishan1522, 2025-01-20:
wait im confused @JacksonElia why do we even parse the json dump if we can access everything we need over nt? Like am i stupid or are all the nt entries also the part of the json dump? If so like what is the point of parsing the json dump??  Do we need nt to parse the dump? Do we need to parse the dump to access nt? lmao i feel like im confusing myself lol

Like i know some teams altogether don't use limelightlib and just use nt (i.e. 254).

[nt link](https://docs.limelightvision.io/docs/docs-limelight/apis/complete-networktables-api)  [json link](https://docs.limelightvision.io/docs/docs-limelight/apis/json-results-specification)

Ishan1522, 2025-01-20:
> The alternative to this would be figuring out what .avgTagDist returns when it doesn't see any april tags and make it return Double.MAX_VALUE instead.

Didn't u pr this code lol 

[image]

Ishan1522, 2025-01-20:
>  I would rather make it based on network table entries so we don't run into this problem again

What problem?

JacksonElia, 2025-01-21:
Okay so I assume you already know most of this, but I'm gonna explain it really simply so anyone can follow along.

A limelight communicates with our robots over its network via network tables. Here is a list of everything it sends:
https://docs.limelightvision.io/docs/docs-limelight/apis/complete-networktables-api

There are a lot of things like `tx` and `ty` which are just numbers and so reading them from the network tables is very quick and easy.  One of these entries is the json. This is just a string that follows json formatting. So the json is sent over via network tables. This json contains basically all the pose estimation data the limelight gets.

However, actually parsing this json is a lot more expensive than just reading a number and so we need to be mindful of when we do it. Limelight helpers has methods that read values from the network tables, but its main use is the methods that parse the json for you.

Both of the methods we use for getting MT1 (`LimelightHelpers.getBotPoseEstimate_wpiBlue`) and MT2 (`LimelightHelpers.getBotPoseEstimate_wpiBlue_MegaTag2`) parse this json everytime we call them. Therefore, we need to try and minimize when we call these methods.

Ishan1522, 2025-01-21:
Ok but isn't it calling a network table value here (where it calls megatag2?): 

[image]

The only place where the json is dumped is in LimelightResults (what we used to use way back in 2024), I don't believe it is accessed by the PoseEstimate (which does it's own nt accessing and stuff) object which we use now...? 

Here is limelight results: 

[image]

JacksonElia, 2025-01-21:
Ah shit I looked into it and you're totally right. I have just been thinking about this the wrong way forever...

That being said, I don't think accessing an entry in an NT is completely negligible so I guess we should still try to reduce it, but it really doesn't matter as much as I thought. 

Damn that's crazy

Ishan1522, 2025-01-21:
We can replace the getNumberOfAprilTags with the tv thingy from nt or add it in idk

closed 2025-02-01 by Ishan1522