TitaniumTigers4829/mako-robot-code-2025 #38: Vision filters

Ishan1522, 2025-01-18:
- [ ] Teleportation filter - don't include measurement if pose is flying around
- [ ] Only include measurement if poses are close together
- [ ] ambiguity - idk prolly would help

JacksonElia, 2025-01-18:
What do you mean by the second one? Do you mean if MT1 and MT2 measurements are close together, or if multiple LL measurements are close together? If it's the first one, idk if we want to do that because MT2 can be really fucky sometimes even when MT1 is decently accurate (like if you're spinning fast)

Ishan1522, 2025-01-18:
its meant to be if multiple limelight measurements are close together

Ishan1522, 2025-01-21:
Some filtering examples: [3015](https://github.com/3015RangerRobotics/2024Public/blob/main/RobotCode2024/src/main/java/frc/robot/subsystems/localization/Localization.java) & [254](https://github.com/Team254/FRC-2024-Public/blob/main/src/main/java/com/team254/frc2024/subsystems/vision/VisionSubsystem.java) <- both these teams are gud.

I might add some more (i think 2910 might use limelights for april tags? 6328 also has an example proyecto)
Something else to think about is what we do after the data is filtered.

closed 2025-02-01 by Ishan1522