TitaniumTigers4829/mako-robot-code-2025 #62: write unit tests

Ishan1522, 2025-02-02:


JacksonElia, 2025-02-02:
Do we use Junit? I can help with writing unit tests, but I'm not exactly sure how to write them. Is there any support for making like fake network table entries and then manually calling .periodic and stuff?

Ishan1522, 2025-02-02:
> Do we use Junit?

yes

>  Is there any support for making like fake network table entries

yes https://docs.wpilib.org/en/stable/docs/software/networktables/multiple-instances.html#networktables-instances

> manually calling .periodic and stuff?

probably? u can use the HAL & simulation in unit tests so like u could probably do that. idk if we will want to do that tho? we don't  do much vision stuff in periodic other than updating inputs.

JacksonElia, 2025-02-03:
okai i might make a pr for this

JacksonElia, 2025-02-03:
wait actually @Ishan1522 do you want me to? I've been trying not to actually do much of the coding myself cause I'm a mentor, but no one wants to do unit tests so idk

Ishan1522, 2025-02-03:
High key if your coming to the meeting today probably teach someone how to write them

Ishan1522, 2025-02-03:
Or maybe not idk they gotta finish their subsystems. But I'm fine with u writing stuff as long as u document it. I can also write some tests 😁

PR #102 (2025-02-25): Add Vision tests and switch to TigerHelpers
This was actually awful to get working, the solution ended up being really simple but going off of these docs was not it...
https://first.wpi.edu/wpilib/allwpilib/docs/release/java/edu/wpi/first/networktables/NetworkTableInstance.html

I only did a really basic test as an example, but any help is appreciated. If anyone is interested in testing/wants to learn why/how we do it, let me know!!

Also, I (chat gpt mostly) made a github action for the tests, and it has to build the code to run the tests, so @Ishan1522 what do you think about removing our ac [... cut]
PR #101 (2025-02-25): devbot 🙏 
Closes #84 
Closes #62
closed 2025-02-25 by Ishan1522