TitaniumTigers4829/mako-robot-code-2025 #65: Cleanup the code

Ishan1522, 2025-02-03:
There's a lot of little things in the code that annoy me right now

- [ ] The devbot thing that doesn't even work (and the switch statement that isn't created correctly)
- [ ] extras is kind of a mess
- [ ] RobotContainer is a mess (maybe can be merged into Robot??)
- [ ] characterization should be in commands, not extras :/ idk what I was thinking with that one.

JacksonElia, 2025-02-04:
Wait wdym merging RobotContainer into Robot?

Ishan1522, 2025-02-04:
it's literally a container of shit that goes into robot and gets run and sometimes i feel like i have to work around that. idk if it'd be cleaner it's just an idea

JacksonElia, 2025-02-04:
I agree that it def could be cleaned up, but I think we should keep it because it kinda stops us from accidentally fucking up important stuff in robot, but more importantly is that it is kinda FRC standard and part of the CommandBased framework. A Timed robot doesn't have robot container

Ishan1522, 2025-02-04:
> up important stuff in robot

Like what? There are just a couple more methods in Robot. This "merging" would probably benefit us in that we would gain access to those methods instead of limiting ourselves to teleopInit and autonomousInit in RobotContainer. And currently if we did want to use the methods in robot we would have to instantiate subsystems that already exist in RobotContainer.

> FRC standard and part of the CommandBased framework

Since when? Have you researched this thoroughly, other than the base robot skeleton that gets created, what is recommended by Wpilib developers on the Wpilib GitHub and the actual updated templates. I implore you to look into it.

> A Timed robot doesn't have robot container

Ok? Who cares? A robot container is not what makes something a timed robot project. It's how you actually code the project. Since we use commands we are command-based. A RobotContainer just contains things that go in Robot.

Ishan1522, 2025-02-05:
maaaaaaaaaaaaaaaaaaaaaan whatever this shit is not even that big of a deal

Ishan1522, 2025-02-05:
just please look into best practices a little more.

i don't really give a shit about the robotcontainer thing

Ishan1522, 2025-02-05:
here is what you get when you search "robotcontainer" in the wpilib pr thingy. https://github.com/wpilibsuite/allwpilib/pulls?q=is%3Apr+is%3Aopen+robotcontainer

JacksonElia, 2025-02-05:
Really? I mean I feel like the majority of teams that use CommandBased have robot container

https://github.com/Team-4795/2025-Reefscape/tree/main/src/main/java/frc/robot
https://github.com/Team254/FRC-2024-Public/tree/main/src/main/java/com/team254/frc2024
https://github.com/Mechanical-Advantage/RobotCode2025Public/tree/main/src/main/java/org/littletonrobotics/frc2025
https://github.com/FRCTeam2910/2024CompetitionRobot-Public/tree/main/src/main/java/frc/robot

I mean I could keep going, but I looked up 4 random teams and all of them had it. Like I agree this argument doesn't really matter, but I'm inclined to keep it in our code because it puts us more in line with the rest of FRC. Like if another team looks at our code, or maybe we need help from a CSA they'll get whats going on.

Ishan1522, 2025-02-05:
> its just what I'm used to but I feel like it nicely keeps our code modular

It's unnecessary modularity tho.

> Why would they include it if they didn't recommend it lol

they are volunteer with lives? And main dev effort is focused towards 2027 updates now. did you click on the link i sent? It is planned to remove RobotContainer entirely by 2027 (the pr is literally approved), when the new controller comes out. There has been a pr open to remove it for a year.

>  I looked up 4 random teams and all of them had it

? ok i can find teams who only have robot so like what's your point here?

https://github.com/frc6995/Robot-2025/tree/main/src/main/java/frc/robot
https://github.com/CurtinFRC/2025-Reefscape/tree/master/src/main/java/org/curtinfrc/frc2025
https://github.com/McQuaidRobotics/2024_Crescendo/tree/main/src/main/java/com/igknighters
https://github.com/icrobotics-team167/2025_Reefscape/tree/main/src/main/java/frc/cotc

i could keep going.

> Like if another team looks at our code, or maybe we need help from a CSA they'll get whats going on.

the content of the code is the same tho?

Ishan1522, 2025-02-05:
>  the rest of FRC.

the rest of FRC will be in for a shock when the recommendation is to switch in 2027 (and RobotContainer is wiped out from all the templates). I say we should be ahead of the game and switch now. 1. for cleaner code, and 2. so we don't have to worry about it later.

Ishan1522, 2025-02-05:
> the difference is you probably had to cherry pick those teams.

The difference is that these are teams I know with people that contribute to Wpilib.

Also when did we start using East as a code example 💀💀

> It's 2025, not 2027

This is not an excuse. We should constantly be pushing to improve our code and future proof it. Staying stuck in the past with deprecated features or features that are not recommended is not something I value.

> just be nicer about it 

I apologize, I appreciate your opinion, however you opinion tends to be a couple years out of date and I haven't seen you show much effort to change that, hence why I pushed so hard for this. And for you to actually research why I was suggesting this, which you didn't do.

JacksonElia, 2025-02-05:
> The difference is that these are teams I know with people that contribute to Wpilib.

Okay fire point. If you want to do this, let's do it now rather than once we have a bunch of subsystems and button bindings

Ishan1522, 2025-02-07:
Lmao https://www.chiefdelphi.com/t/simple-timed-based-subsystem-to-command-based-code-example/485479/3?u=ishan1

closed 2025-02-17 by Ishan1522