TitaniumTigers4829/mako-robot-code-2025 #69: Consider a superstructure type thing

Ishan1522, 2025-02-04:


JacksonElia, 2025-02-04:
no

JacksonElia, 2025-02-04:
lol that was mean reopen it if u want

Ishan1522, 2025-02-04:
> no

Explain.

JacksonElia, 2025-02-04:
link an example of what you mean

Ishan1522, 2025-02-04:
[6328](https://github.com/Mechanical-Advantage/RobotCode2024/blob/main/src/main/java/org/littletonrobotics/frc2024/subsystems/superstructure/Superstructure.java)
[4788](https://github.com/CurtinFRC/2024-Offseason/blob/master/src/main/java/frc/robot/Superstructure.java)
[2910](https://github.com/FRCTeam2910/2024CompetitionRobot-Public/blob/main/src/main/java/frc/robot/subsystems/Superstructure.java)

[their readme is also sick](https://github.com/FRCTeam2910/2024CompetitionRobot-Public/tree/main?tab=readme-ov-file#organization)

Ishan1522, 2025-02-04:
yo i know i keep bringing this idea up @JacksonElia but it's because i think it'd be cool and clean. im not super pressed about adding it into the code

JacksonElia, 2025-02-04:
Idk I still just don't love it. I feel like a CommandBased robot is meant to be modular and I don't love 1000 line (with like no comments) files. 

You are head of programming tho so when it comes down to it, it's your decision. 

>Explain

I mainly just feel like its cluttered and we can do what it does by just writing better commands. 2910's has multiple 100 line switch statements

Ishan1522, 2025-02-04:
Ok I don't rlly care

closed 2025-02-04 by Ishan1522