TitaniumTigers4829/mako-robot-code-2025 #125: Better drivetrain controls

Ishan1522, 2025-03-04:
Tbh wheel slip might actually be important bc our carpet hella slippy (setpoint gen is a preventative measure against slipping for swerve modules, if carpet is shit we need another solution). We might want accel limiters and rotation control on drivetrain. This can be a comp 1 thing if time allows bc it's pretty easy.

Ishan1522, 2025-03-04:
bro what

JacksonElia, 2025-03-05:
I thought you said you got rid of it but I guess not. If wheel slip is still a problem, can we tweak some of the SSP parameters to help? Also why does it have so many magic numbers also I think u need to change the friction coefficient now
```java
private final SwerveSetpointGenerator setpointGenerator =
      new SwerveSetpointGenerator(
          DriveConstants.MODULE_TRANSLATIONS,
          DCMotor.getFalcon500(1).withReduction(ModuleConstants.DRIVE_GEAR_RATIO),
          DCMotor.getFalcon500(1).withReduction(ModuleConstants.TURN_GEAR_RATIO),
          60,
          58,
          7,
          ModuleConstants.WHEEL_DIAMETER_METERS,
          WheelCof.BLACK_NITRILE.cof,
          0.0);
```
Wheels arent black nitril anymore

Ishan1522, 2025-03-05:
drive current limit, robot weight kg, moi, torque loss

Ishan1522, 2025-03-05:
> you got rid of it but I guess not.

i did. there is a method that allows us to use the simple setpoint gen which we are currently using.