TitaniumTigers4829/mako-robot-code-2025 #137: make auto align work during robot motion

Ishan1522, 2025-03-10:
the transition from a robot state of moving to auto align is currently not easy for the code to handle (maybe need a profile, but that is probbably not the only solution). so we had to do auto align from a stopped state for it to align accurately.

JacksonElia, 2025-03-12:
Yeah I'm not sure the best way to do this. Maybe a well tuned profile PID controller, but like u said we probably need to do something else

aridavidson001, 2025-03-14:
Could we use pathplanner to do real-time trajectory generation like outlined [here](https://docs.google.com/document/d/10if4xjAaETTceUVn7l4J-jOCOnm5CJUDS5RAVNIJMQM/edit?tab=t.0)?

Ishan1522, 2025-03-14:
write code that uses it and figure out if it works

closed 2025-03-20 by Ishan1522